bhagman/Tone #8: Tone messing with delay?

Darenn, 2018-10-23:
Hi!
I'm noob at Arduino, so I might be the wrong.
Is it possible that the library mess with the _delay()_ function?
It seems that after using _play()_ on different tones, all my following calls to the _delay_ function are very slower.

bhagman, 2018-10-23:
What board are you using and how many tones are you creating?

Darenn, 2018-10-23:
Thanks for your reply.
I'm using an Arduino ELEGOO  UNO R3.
I'm creating two tones, it's does not work at all when trying to create more than 3 tones.

bhagman, 2018-10-23:
Unfortunately, the ATmega328P (on the UNO R3) can only support up to 2 tones without interfering with the "millis" timer (which is used for `delay()`).

You can find the details here: https://github.com/bhagman/Tone#ugly-details

Darenn, 2018-10-23:
Oh I understand better now thanks. 
But even when I'm using two tones it seems to interfere with delay(). According to the readme it shouldn't be the case. I thought that maybe timer 0 was used first, but not...
I certainly made a mistake somewhere else, I'll keep looking.
Thanks for your help!

happysupply, 2019-01-08:
Dear @bhagman, many thanks for your non-blocking Tone Library, it helps me tremendously. A follow-up question on this issue posted by @Darenn.

So if I'm producing only one tone simultaneously, I'm only using hardware timer 2 with the Tone Library. This means that timer 0 is left untouched by the Tone Library, so millis() will run equally smooth as compared to a situation in which no tones are produced. Correct?

nathanRamaNoodles, 2019-02-05:
If anyone wants a way to play infinite tones at the same time without blocking the main program, then use [Music Without Delay](https://github.com/nathanRamaNoodles/MusicWithoutDelay-LIbrary).

bhagman, 2019-02-05:
@happysupply yes, if you are only using Timer 2, Timer 0 will not be altered by the Tone library.

A few people have asked about having a way to use 3 tones at the same time (on the '328 parts) -- and I have an uncommitted change that will allow using any of the tones as a "delay" substitute -- so you can use 3 tones AND be able to have regular delays.

@nathanRamaNoodles --- awesome!  I have seen a few of the PWM tone substitutes around, putting Tone to shame :)  I'm going to go over to your repo and make some suggestions.

closed 2019-02-05 by bhagman

nathanRamaNoodles, 2019-02-06:
I wouldn't say "put to shame", it's rather more of an inspiration.  Your Tone library was the first one I used to run MusicWithoutDelay. I liked how you could write music easily with RTTL.  But then I ran into the issue of polyphony, and blocking the main program with delay().
 
   Eventually, I learned about timer registers and wavetable synthesis.  And then I discovered volume control on my own(which still surprises me to this day).  Volume control on with DAC is more difficult than i2s.

Suggestions are cool too.  I made a MIDI piano with the library, but I haven't released it yet.