bhagman/Tone #11: multiple definitions of timer0_pin_port and others

MatersM, 2020-02-20:
Folk,

I was afraid that this library was no longer support, but I do see activity here, so I have good hopes. Trying to use the library with Arduino 1.8.9 with an Arduino Uno, but can not get the library to work. See for more information below. Help will be much appreciated.

Regards,

Martin

Arduino:1.8.9 (Mac OS X), Board:"Arduino/Genuino Uno"
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer0_pin_port'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer0_pin_mask'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer1_pin_port'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer1_pin_mask'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer2_pin_port'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer2_pin_mask'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer0_toggle_count'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer1_toggle_count'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `timer2_toggle_count'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `__vector_7'
libraries/Tone/Tone.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Fout bij het compileren voor board Arduino/Genuino Uno

bhagman, 2020-02-24:
Hey Martin,

Yes -- very slowly.  Thanks for the patience.

I'll have a look and see what I can do.

Thanks!

MatersM, 2020-02-24:
Hey Brett,

That would be great.
I really would like to use your library.
Currently using the build-in ‘tone’, but actually multiple tones is what I’d like to use.

I’ll try to be patient :-)

Best regards,
Martin

Verstuurd vanaf mijn iPhone

bhagman, 2020-02-25:
Are you installing the library through the Arduino IDE Library Manager?

MatersM, 2020-02-25:
Nope, via a ZIP file.
Did try it with the Library Manager, but couldn’t find it.

Verstuurd vanaf mijn iPhone

bhagman, 2020-02-25:
I'm not sure where you got the ZIP file from.  I'll wager it was from the very old Google Code repo.

If you use the Arduino IDE Library Manager (Sketch -> Include Library -> Manage Libraries...) :

[image]

and use 'tone generation' in the search, you will find the latest version of the library:

[image]

Try that out and let me know how it goes.

MatersM, 2020-02-25:
Hi Brett,

Some good/bad/good news:
- I found the library and used it, with the same result, however…
- I’m using another library <Timer.h> as well and that seems to interfere in some way with <Tone.h>, I’m using this Library for some timings around the music I’m playing and for some other events.
- After removing <Timer.h> from the includes <Tone.h> seems to work.
- Then I included <Timer.h> again and it still seems to work (I didn’t use Tone yet, but at least the errors where gone)
Strange though…. 

I’ll do some more testing tonight.
Anyway, thanks for your prompt support, much appreciated!

Best regards,

Martin

bhagman, 2020-02-25:
OK, good to hear -- let me know how it goes.

I had a look at the Timer library (https://github.com/JChristensen/Timer).  It only depends on millis()`, so it should work with the Tone library -- unless you hit the last timer when creating Tones.

MatersM, 2020-02-26:
Seems to work fine now.
Once again thanks for the support.

bhagman, 2020-02-26:
Great!  If you can, share what you've done!

closed 2020-02-26 by bhagman

MatersM, 2020-02-27:
Hi Brett,

As said before:
- I deleted the include for <Timer.h>
- Compiled it again (with errors because <Timer.h> was not included.
- Then included <Timer.h> again and it worked like a charm.

Regards,

Martin