bhagman/Tone #13: 20000Hz tone

JlnWntr, 2020-03-15:
If you do this:
`tone1.play(20000);`
you should actually notice/hear something that is barely audible. But this generates some weird raspy sound no matter what speaker is used.
Is it even possible to generate such a tone with this library and hardware (Arduino)?

AuxiliaryPanther, 2022-05-19:
Did you use a tone generator to determine that you can physically hear 20 KHz? I can only hear up to about 17 KHz myself. I noticed a similar sound, too. Ramp the frequency up from 5 KHz to see how high you can hear that speaker. I am not super educated in frequencies, but you could be hearing a constructive combination of harmonics that is within your hearing range.

JlnWntr, 2022-05-24:
I haven't thought of this! Thanks for the reply.

closed 2022-05-24 by JlnWntr