bhagman/Tone #17: Compile for Leonardo?

JeremySCook, 2021-09-16:
Hi, I can get an example file to compile for an Uno, but not for a Leonardo (Pro Micro).

Would it be possible to add this functionality, or perhaps I'm missing something? Using 1.1.13 if that helps.

JeremySCook, 2021-09-17:
So thanks to [@diyelectromusic ](https://twitter.com/diyelectromusic/status/1438570910949289996)on Twitter, the included .txt file (with a .cpp extension instead of .txt) can be subbed to have it compile for Leonardo (tested on a Pro Micro). Seems to compile for an Uno as well, but have not actually tested it there.

[Tone-leo.txt](https://github.com/bhagman/Tone/files/7181641/Tone-leo.txt)

diyelectromusic, 2021-09-22:
I now have a version of this library that can support all four timers on the ATmega32U4 giving four tones on an Arduino Leonardo. Is that something of interest?

I'm a bit of a github newbie, but if you are interested, I'll attempt to submit a PR!

I have some preliminary tests on an Uno and Pro Micro, but I haven't tested it on the other supported boards or been able to test it particularly exhaustively.  I do have oscilloscope traces showing that the timings appear correct for all four timers for a limited range of frequencies.

Kevin