bhagman/Tone #18: No tone, or "rest"

pdr0663, 2021-11-04:
I am using your library to score music rendered as chords, and Ineed all three timers on an ATMEGA 328 (Arduino Nano). Problem is, as all timers are occupied, what do I do when I want to rest? I can't seem t ouse delay() any more.

I use 20000Hz for rests, ut I get some clicking artifacts.

Can you add code to deal with perhaps "0" frequency, which would correspond to "no sound", for the specified tone() period? That way I can use the tone() command for sounds and rests.

Thanks for the great library!

Paul

AuxiliaryPanther, 2022-05-19:
Brute force idea: use a function that effectively does nothing, like changing digital level on an unused pin, time it, and loop that as long as you need to achieve the delay. E.g.:

void bruteDelay(int iterations){
  For (int i, i <= iterations, I++){
    digitalWrite(unusedPin, !digitalRead(unusedPin));
  }
}
bruteDelay(200); //delays 200 cycles

Blargian, 2023-12-21:
> I am using your library to score music rendered as chords, and I need all three timers on an ATMEGA 328 (Arduino Nano). Problem is, as all timers are occupied, what do I do when I want to rest? I can't seem to use delay() any more.
> 
> I use 20000Hz for rests, but I get some clicking artifacts.
> 
> Can you add code to deal with perhaps "0" frequency, which would correspond to "no sound", for the specified tone() period? That way I can use the tone() command for sounds and rests.
> 
> Thanks for the great library!
> 
> Paul

You could use [noTone](https://www.arduino.cc/reference/en/language/functions/advanced-io/notone/) for that I believe.